DavesCodeMusings/mpremote-vscode #1: Does not work under linux

petrkr, 2023-05-16:
Seems this module think everybody at the whole world using windows, so this extension will NOT work in Linux nor Mac Os

```
$ py.exe -m mpremote devs
bash: py.exe: command not found
```

Please make it platform independent by using python3 command, it should be available on every platform in PATH

DavesCodeMusings, 2023-05-16:
Unfortunately, `python3` did not work on Windows, but `py` does.

```
PS C:\> python3 --version
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
PS C:\> py --version
Python 3.11.0
```

So, I think this is not a simple find and replace fix. I suppose there may be a way to determine the underlying OS and adjust the executable name to suit, but I'm not sure what it is at the moment. If you have any ideas or sample code for doing that, let me know.

petrkr, 2023-05-16:
Seems like wrong python framework is installed in windows, there normally is python executable, do not know what kind of bundle you installed. But if you will use regular python it has python executable.

I do not have windows right now, so I can not test official release from https://www.python.org/downloads/windows/ but my college using windows and he is using python executable for sure

DavesCodeMusings, 2023-05-16:
Pretty sure it was the download from https://www.python.org/ Interestingly, I get different versions with `py` and `python` and there's no `python3`.

```
PS C:\> py --version
Python 3.11.0
PS C:\> python --version
Python 3.9.2
PS C:\> python3 --version
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
```

petrkr, 2023-05-16:
So maybe windows missing symlink/whateverlink on python3 from python.

But since python 2.x is deprecated since 2021, it can be used python without specify version at all... And if someone still have default python 2.x as python command, they probably using out-of-dated OS

DavesCodeMusings, 2023-05-16:
I removed and reinstalled Python using the defaults given by the installer. Now only `py` works.

```
PS C:\> py --version
Python 3.11.3
PS C:\> python --version
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
PS C:\> python3 --version
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
```

I think I need to dig into this further.

DavesCodeMusings, 2023-05-17:
Tried a Linux host and got the following results:

```
alpine:~$ py --version
-ash: py: not found
alpine:~$ python --version
Python 3.10.8
alpine:~$ python3 --version
Python 3.10.8
alpine:~$ ls -l `which python`
lrwxrwxrwx    1 root     root            16 Dec  8 20:23 /usr/bin/python -> /usr/bin/python3*
alpine:~$ ls -l `which python3`
lrwxrwxrwx    1 root     root            10 Dec  8 20:23 /usr/bin/python3 -> python3.10*
```

It looks like `python` is the most generic way to call the Python binary on Linux. I don't know if this is the same behavior you'd get on all flavors of Linux/MacOS/BSD or not, but it's a starting point.

So I'm thinking of this approach...
1. Detect if it's Windows or other using `os.platform`
2. If it detects Windows, set the variable `PYTHON_BIN = "py.exe"`
3. If not windows, set the variable `PYTHON_BIN = "python"`
4. Try running `PYTHON_BIN --version` to verify.

petrkr, 2023-05-18:
I installed windows 10 under VirtualBox in order to test python there. I downloaded official windows installer from python.org. Checked "Add python to PATH" before install and then both py and python commands it working fine.

I thing best solution is to use "python" command, because it works everywhere

[image]

petrkr, 2023-05-18:
Also tried pip to install mpremote

[image]

DavesCodeMusings, 2023-05-19:
I added a few lines of code to check the OS type when the extension activates. If it detects Windows, it'll use `py.exe`. For everything else, it will use `python`.

I [committed the code](
https://github.com/DavesCodeMusings/mpremote-vscode/commit/f3318d668d62d10ed4cda58a74ce1cb037aa6a6a), but did not publish the extension yet. I was only able to test it on Windows and verify the change does not break anything, but I don't have a VS Code instance on Linux to test against.

If you want to clone the repo and test it in VS Code's debugger on Linux, that would be great. Let me know how it goes.

petrkr, 2023-05-20:
I never installed extension from code. I tried to check out this repo to exptensions folder and run npm install, but it seems it does not work (not see that plugin). Maybe would be faster if you have some link how to do that.  Problem of VS code and google is if you search something like "extension from github" or "from source codes" it will find extensions which are working with github or git/svn/etc..

DavesCodeMusings, 2023-05-20:
@petrkr No problem. I can just publish it. The changes don't break anything for VS Code on Windows.

DavesCodeMusings, 2023-05-20:
@petrkr Published changes as version 0.4.4. Give it a try when you have a chance and let me know how it goes.  https://marketplace.visualstudio.com/items?itemName=DavesCodeMusings.mpremote

DavesCodeMusings, 2023-05-24:
I'm assuming this will work, but I don't have a Linux host with VS Code to test on. The changes didn't break anything in Windows VS Code, so I'm calling it good and closing the issue.

closed 2023-05-24 by DavesCodeMusings