DavesCodeMusings/mpremote-vscode #2: Run command cannot handle path/filename with space(s)

erikdelange, 2023-05-20:
Situation: Filename in the active editor is _c:\Directory with a space\hello.py_

Executing command _MPRemote: Run File Shown in the Active Editor (run)_ will fail with the following output:

> (MicroPython) PS C:\Directory with a space> py.exe -m mpremote connect COM6 run c:\Directory with a space\hello.py
mpremote: could not read file 'c:\Directory'

Problem is that the filename is truncated at the first space. 
When manually adding quotes around the file argument and rerunning the command it does work.

> (MicroPython) PS C:\Directory with a space> py.exe -m mpremote connect COM6 run "c:\Directory with a space\hello.py"
Hello there

Using:
- MPRemote v0.4.3
- Windows 11
- Python 3.11 (with a venv called MicroPython hence the name of the prompt)
- mpremote 1.20.0

DavesCodeMusings, 2023-05-20:
I'll take a look at adding quotes. Thanks for posting the issue with a proposed solution.

DavesCodeMusings, 2023-05-20:
Added single quotes around paths used with the run and copy commands to allow for spaces. Tested okay on my Windows installation. Pushed changes to GitHub and published to VSCode Marketplace.

@erikdelange Please test when you have time and let me know if it works for you.

erikdelange, 2023-05-20:
Tested and it works perfectly. Thank you for fixing this so quickly.

closed 2023-05-20 by erikdelange