DavesCodeMusings/mpremote-vscode #5: Ability to ignore certain COM ports

erikdelange, 2023-06-10:
A handy feature of mpremote is that when there is only one MicroPython board attached the user does not need to select a COM port before a command is executed.

However, I'm also using COM ports for other purposes. As a consequence when executing an mpremote command I always have to select the COM port my board is attached to, even if I have only one board connected. It would be a nice feature if - via a configuration key - a user can record COM ports which should be discarded.

(Would somehow have to be done (in function getDevicePort()).

As an example the COM ports I get when running 'Scan for attached devices'. COM6 is a MicroPython board, but COM1 is something completely different.

```
COM1 None 0000:0000 (Standaardpoorttypen) None
COM6 DQ008V27A 0403:6015 FTDI None
```

DavesCodeMusings, 2023-06-11:
Note to self: https://code.visualstudio.com/api/references/contribution-points#contributes.configuration

DavesCodeMusings, 2023-06-13:
@erikdelange How does this look for a start? Does the description make sense? This is only the config setting. I haven't added code to `getDevicePort()` yet, but it should just be a matter of checking detected ports against this list and splicing them out of the array.

[image]

erikdelange, 2023-06-13:
Looks perfect! Very clear description.

DavesCodeMusings, 2023-06-13:
Added in 1.20.8

closed 2023-06-13 by DavesCodeMusings