DavesCodeMusings/mpremote-vscode #6: MPRemote ignores almost every command from pallete

petrkr, 2023-06-13:
I trying to use mpremote in VSCode OSS under linux, but most of options seems to do nothing, even no error messages in developer debug window. It act as no-action binds..

for example show devices will open terminal and run devices, but others like repl, mkdir, sync... etc, nothing.

Is there any way how to put it to more debug mode to check whats going on ?

DavesCodeMusings, 2023-06-14:
I think to see debug output you'll need to clone the extension's GitHub repository and run it using the VS Code debugger. That's how I do it, but I'm developing on a Windows system, so obviously I won't see exactly what you're seeing. However, there's nothing really magical about the extension. It's just putting together mpremote command-line options and running the command in the VS Code terminal window. Are you seeing any errors in the terminal window or is it just nothing coming up?

ccoupe, 2023-09-08:
I ran into this problem too. I am also running Linux. The mpremote commands that need an additional input - `chdir` for example do not bring up the dialog to enter the rest of the command line. However some commands do bring up the dialog - mkdir for example. I don't see any errors in the terminal window. I can type in the mpremote command sin that window so the shell connection is OK.

DavesCodeMusings, 2023-10-16:
I don't use VS Code in Linux, so I have no way of confirming or testing. If anyone with a Linux installation of VS Code and knowledge of VS Code extensions wants to help out, I'd appreciate it.

ccoupe, 2023-10-16:
I'd help but I don't want to learn .ts (typescript I suspect) and how to debug it and how to use it inside VSCode. You could install virtualbox and a smallish linux on Windows or OSX and install VSCode in that Linux. Or as I mentioned install Windows Subsystem for Linux (WSL).

DavesCodeMusings, 2023-10-16:
I'm kind of in the same boat. My primary desktop OS is Windows, so that's what I develop and test on. I have no desire to install VS Code on Linux. Not trying to be flippant, it's just that my time is limited. If someone else can do this, great. That's why I'm leaving the issue open and tagged it with help wanted.

ccoupe, 2023-10-17:
Looks like 'functions', if that is the correct term, looks more like a block to me,  that use vscode.window.showQuickPick() have this problem - see extension.ts.  That seems like a well used function in VScode so I expect it works OK on Linux.

ccoupe, 2023-11-04:
For what it is worth, the same failure occurs with OSX - it's not just Linux.

DavesCodeMusings, 2023-11-04:
Out of curiosity, what happens if you create a directory and then remove it? For example:

```
MPRemote: Make a new directory (mkdir)
MPRemote: Remove a directory from remote filesystem (rmdir)
```

You should see an InputBox asking for the directory name when you do `mkdir` and then a QuickPick showing you choices for directories to deleted when you do the `rmdir`.

Do you get these prompts?

ccoupe, 2023-11-04:
```
ccoupe@bronco:~/Projects/iot/micropython/esp32-s3/spidisplay/src$ 
ccoupe@bronco:~/Projects/iot/micropython/esp32-s3/spidisplay/src$ python -m mpremote connect /dev/ttyACM1 fs mkdir '/foo'
mkdir :/foo
ccoupe@bronco:~/Projects/iot/micropython/esp32-s3/spidisplay/src$ python -m mpremote connect /dev/ttyACM1 fs ls '/'
ls :/
           9 boot.py
        3139 config.py
       80675 deja48.py
        2164 font_test.py
           0 foo/
        1936 main.py
         291 memchk.py
        9269 messagedevice.py
       29113 mqtt_as.py
         203 rangerdevice.py
         224 scan.py
       36743 st7789py.py
```
There is no prompt for the rmdir and no drop down list. 

My guess is the list is empty (bug) and vscode skips the prompt (bug?) or the Quickpick is incorrect on Linux and OSX. There are hints in the massive github issues list that there are two picking mechanisms and they can't be mixed. I don't know if that apply here.

DavesCodeMusings, 2023-11-04:
In reference to: "There are hints in the massive github issues list that there are two picking mechanisms and they can't be mixed. I don't know if that apply here."

Are there any links you can provide? My searching has not come up with anything yet.

ccoupe, 2023-11-04:
Try google search  `github: quickpick` seems to restrict the query to github. That makes it easier. One thought I had was to find another vscode extension that has drop down quickpick and see if it's broken in Linux  - if not broken then it would have clues to something that does work.

DavesCodeMusings, 2023-11-04:
Since it looks like you're using an ESP32, I'm wondering if you'd be willing to try out my esptool extension for VS Code on a spare microcontroller you don't mind flashing. There is a QuickPick prompt when you flash firmware that asks for the offset address (0x0 or 0x1000). I'd be curious to know if the QuickPick in esptool works. The way I've coded the two extensions is nearly identical. But, the choices offered by esptool for 0x0 or 0x1000 are hard coded, whereas with MPRemote, the directory choices are gathered from a short python script running on the microcontroller. If the QuickPick offered by esptool works, but not MPRemote, I would say the problem is not with QuickPick, but the dynamic list of choices. Hopefully that makes sense.

DavesCodeMusings, 2023-11-04:
Okay, I think I found it. This is the short bit of code I run to find directories and files on the microcontroller...

```
        let oneLiner = `from os import listdir, stat ; print([entry for entry in listdir('${cwd}') if stat('${cwd}' + '/' + entry)[0] & ${mask} != 0])`;

        // TODO: fix hard-coded py.exe
        let listDirCmd = `py.exe -m mpremote connect ${port} exec "${oneLiner}"`;
```

Notice the TODO note I left myself. So yeah, that totally explains why it only works on Windows.

Now that I know what it is, I can work on a fix.  Thanks @ccoupe for the help in narrowing this down.

ccoupe, 2023-11-04:
I have a phrase I use when I'm bug hunting in my code and find one of my TODO comments. God D*mn it, Not Again!

DavesCodeMusings, 2023-11-04:
Mine was something along the lines of Homer Simpson's DOH! I also have a flat spot on my forehead where I smacked myself.

ccoupe, 2023-11-04:
FWIW erase flash didn't ask for an offset but it did vigorously flash the LED on the S3 - but nothing was erased. I think I need the usb cable in the other connector to actually reprogram the thing.

DavesCodeMusings, 2023-11-04:
I just published the fix for the quick pick selections. Once you update to v1.21.3 of the mpremote extension, give it a whirl and see if the selection prompt works for you. Fingers crossed over here.

ccoupe, 2023-11-04:
That was it! Below, I removed the 'foo' dir created earlier and also a file from the drop down list. Thank You. I'll let you close the issue unless you want me too.
```
ccoupe@bronco:~/Projects/iot/micropython/esp32-s3/spidisplay/src$ python -m mpremote connect /dev/ttyACM1 fs rmdir ':/foo'
rmdir :/foo

ccoupe@bronco:~/Projects/iot/micropython/esp32-s3/spidisplay/src$ python -m mpremote connect /dev/ttyACM1 fs rm ':/scan.py'
rm :/scan.py
```

DavesCodeMusings, 2023-11-04:
@ccoupe Excellent news! Thank you for your help and patience in tracking this down.

closed 2023-11-04 by DavesCodeMusings

ccoupe, 2023-11-07:
I hate to do this but there is a related bug in OSX, at least the way my python is setup (homebrew installed). There is no 'python' executable - just python3 - a symlink from python to python3 works well for a short term fix.

DavesCodeMusings, 2023-11-07:
@ccoupe Can you put this under [issue #16](https://github.com/DavesCodeMusings/mpremote-vscode/issues/16)? I'd been wondering about python vs. python3. And #16 will address that.