DavesCodeMusings/mpremote-vscode #8: Error running command mpremote.devs: command 'mpremote.devs' not found. This is likely caused by the extension that contributes mpremote.devs.

kjm1102, 2023-08-07:
When I open VS code it invites me to attach a microcontroller & refresh to access MPremote commands from the context menu. The terminal window displays 
```
PS C:\Users\Admin>
 *  History restored 

PS C:\Users\Admin> 
```
So I click refresh & nothing happens.

DavesCodeMusings, 2023-08-07:
I'm not able to replicate this with v1.20.17 of the extension on Windows 11. Can you tell me what OS and version of the extension you're using?

kjm1102, 2023-08-07:
Version: 1.81.0 (user setup)
Commit: 6445d93c81ebe42c4cbd7a60712e0b17d9463e97
Date: 2023-08-02T12:37:13.485Z
Electron: 22.3.18
ElectronBuildId: 22689846
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Windows_NT x64 10.0.19045

Windows10 home ver 22H2 OS build 190453208

MPRemot 1.20.17

There are at least 2 different versions of python on my windows PC but I thought I saw a fleeting message about python not installed? This probably explains why VS code has never been able to find an esp32 comm port & none of the extensions I've tried (eg pymakr) never work.

DavesCodeMusings, 2023-08-07:
I'm a bit puzzled by Chromium: 108.0.5359.215. Are you using VS Code in a browser? I know it can run that way, but I've never tried it.

kjm1102, 2023-08-08:
The chrome reference puzzled me too because I open it from the start menu so it's not running in the browser.
[image]

kjm1102, 2023-08-08:
Ah caught 'em before they vanished, my 2 transient messages about python problems
[image]

Woazboat, 2023-08-11:
I've run into the same issue on Debian

Version: 1.81.0
Commit: 6445d93c81ebe42c4cbd7a60712e0b17d9463e97
Date: 2023-08-02T12:36:11.334Z
Electron: 22.3.18
ElectronBuildId: 22689846
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Linux x64 6.3.0-1-amd64

This comes up in the vs code extension host logs
```
2023-08-11 23:36:32.290 [error] Activating extension DavesCodeMusings.mpremote failed due to an error:
2023-08-11 23:36:32.290 [error] TypeError: Cannot read properties of null (reading 'replace')
	at PortListDataProvider.refresh (/home/<user>/.vscode/extensions/davescodemusings.mpremote-1.20.17/out/serialportExplorer.js:18:92)
	at process.processTicksAndRejections (node:internal/process/task_queues:96:5)
	at async activate (/home/<user>/.vscode/extensions/davescodemusings.mpremote-1.20.17/out/extension.js:17:5)
	at async p.n (/usr/share/code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:102:6206)
	at async p.m (/usr/share/code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:102:6169)
	at async p.l (/usr/share/code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:102:5626)
```

DavesCodeMusings, 2023-08-11:
From the error provided in @Woazboat 's sample, it looks like it's having a problem with the serial port skip list. Can you check the settings for the extension and see if you have the mpremote settings shown in the screen capture below? It's acting like the setting doesn't exist, but it should be there because it's specified in package.json.

[image]

Woazboat, 2023-08-12:
Yes, the settings exist. If I enter some bogus values into the skip list setting and restart vs code it actually works. (And fails again if I clear the settings field.)

DavesCodeMusings, 2023-08-12:
Good to know. Thanks for the verification.

DavesCodeMusings, 2023-08-12:
I made a change to the serial port skip list setting that should fix this in v1.20.18. (The problem was a null default value for the string versus an empty value.) You might need to set the serial port skip list to some bogus value and then remove it. But after that, it should be fine.

Try it out and be sure to let me know if the problem persists. I've never been able to recreate it on my system.

kjm1102, 2023-08-13:
I get a list of serial ports now that I didn't get before but VS code is still claiming it can't find python & therefore acn't run mpremote as a python module. I can't figure this out since python is definetly in Windows path stmt..

DavesCodeMusings, 2023-08-13:
@kjm1102 it is indeed looking for py.exe. I made the assumption on Windows it's py.exe, because that's what the package from python.org installs in the default configuration.

kjm1102, 2023-08-15:
So I need python.org install as opposed to the one I've got now? Is that the same as installing python as an extension in Vs code?

DavesCodeMusings, 2023-10-16:
Original issue appears to be fixed. Secondary issue seems related to Python, not the extension.

closed 2023-10-16 by DavesCodeMusings