DavesCodeMusings/mpremote-vscode #9: Quoting issues in exec command

Woazboat, 2023-08-12:
Single quotes `'` in commands are not escaped correctly

---

`print('foo')`

`python -m mpremote connect /dev/ttyACM0 exec 'print('foo')'`

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'foo' isn't defined
```

---

`print("foo")`

`python -m mpremote connect /dev/ttyACM0 exec 'print("foo")'`

```
foo
```

DavesCodeMusings, 2023-08-12:
I'm not exactly sure how I can fix this. If I escape the single quotes, there's still an error. The only solution appears to be using double quotes in the `print("foo")` or whatever is being executed.

PS C:\Users\Dave\Desktop\NeoPixel> py.exe -m mpremote connect COM7 exec 'print('foo')'
Traceback (most recent call last):
  File "<stdin>", line 2
SyntaxError: invalid syntax
PS C:\Users\Dave\Desktop\NeoPixel> py.exe -m mpremote connect COM7 exec 'print(\'foo\')'
Traceback (most recent call last):
  File "<stdin>", line 2
SyntaxError: invalid syntax
PS C:\Users\Dave\Desktop\NeoPixel> py.exe -m mpremote connect COM7 exec 'print("foo")'  
foo

DavesCodeMusings, 2023-10-16:
Fixing for one style of quoting breaks it for the other.