DavesCodeMusings/mpremote-vscode #19: Mechanism for controlling which files get sync'd

jeffmakes, 2023-11-23:
Hi,

mpremote-vscode looks like a promising replacement for PyMakr - thanks for releasing it.

I'm having a hard time getting it to work in my context; one which I think is a pretty common development workflow.

I have a VSCode project folder, which contains quite a few dotfiles and dotfolders. My actual code is all in the "src" folder. I would like to be able to press one button (invoking the mpremote sync command) to sync only the stuff in "src". At the moment, "sync" copies the whole file tree (1.9MB). Another problem is that the src folder ends up at "/src" on the target, pretty obviously.

mpyremote currently ignores the .git folder as a special case, which is helpful, but it doesn't help with the more general case. Perhaps you could implement a .mprignore file (like .gitignore) to allow the user to ignore more stuff.

That wouldn't address the challenge of getting myproject/src/main.py to /main.py on the target, though. One way to solve that would be a config file that manages the sync process. It could be as simple as a list of mpremote commands, which would allow the user to easily craft a completely bespoke sync process

Below is my folder structure, which indicates the problem.

Cheers!
Jeff

```
[jeff@shaper myproj-mpy]$ ls -la
total 48
drwxr-xr-x. 6 jeff jeff 4096 Nov 23 19:47 .
drwxr-xr-x. 5 jeff jeff 4096 Nov 23 18:52 ..
-rw-r--r--. 1 jeff jeff   12 Nov 23 19:28 dev-requirements.txt
drwxr-xr-x. 7 jeff jeff 4096 Nov 23 18:58 .git
-rw-r--r--. 1 jeff jeff 2184 Nov 23 19:28 .gitignore
drwxr-xr-x. 3 jeff jeff 4096 Nov 23 19:28 .micropy
-rw-r--r--. 1 jeff jeff  294 Nov 23 19:28 micropy.json
-rw-r--r--. 1 jeff jeff 2597 Nov 23 18:56 pybcdc.inf
-rw-r--r--. 1 jeff jeff    0 Nov 23 21:22 .pylintrc
-rw-r--r--. 1 jeff jeff  295 Nov 23 21:52 pymakr.conf
-rw-r--r--. 1 jeff jeff  526 Nov 23 18:56 README.txt
-rw-r--r--. 1 jeff jeff    0 Nov 23 19:28 requirements.txt
drwxr-xr-x. 2 jeff jeff 4096 Nov 23 19:28 src
drwxr-xr-x. 2 jeff jeff 4096 Nov 23 19:28 .vscode
[jeff@shaper myproj-mpy]$ ls src/
boot.py  main.py
[jeff@shaper myproj-mpy]$ 
```

thomaslr, 2023-11-26:
I second this! Pymakr is not unusable micropico is great but seems only to work on pico. Both of these seemed to use an incremental difference sync approach.

maybe even easier would be to add or change the option to run a file on the remote board using

python3 -m mpremote connect /dev/ttyACM0 mount . run '/home/thomaslr/dev/pi-pico/plant-project/water_sensor.py'

which mounts the host machines filesystem and allows access to all referenced files which the current run file on remote device doesn't if it references files that haven't been uploaded to the board.

Seems pretty fast for what I'm doing so far but how it would behave with 1.9MB of data

DavesCodeMusings, 2023-11-26:
It shouldn't be too difficult to add a configuration option for the extension that specifies the root dir to use with the `sync` command. The ignored files can probably be a comma-separated list in a config option as well. I'll take a look at the code and see what I can do.

I'm trying to limit features done by the extension and rely on `mpremote` to provide the functionality. So that will dictate how much gets implemented. But a different upload dir and configurable skipped files shouldn't be too bad.

DavesCodeMusings, 2023-11-29:
I summed up this enhancement request as a minimum viable product of: "Give me a way to configure a src directory so I don't get a bunch of junk uploaded to my microcontoller's flash."

If you install v1.21.7 of the extension and go into its settings, you will find _Mpremote: Src Subdirectory_. 

I have described the setting as:

"If all your MicroPython files are contained in a subdirectory of your project folder (e.g. src), enter that subdirectory name here. This will be used as the local path for uploads to your microcontroller."

Hopefully this all makes sense. It should allow folks to isolate their MicroPython code in one subdirectory and leave all the ancillary files in the project root.

closed 2023-11-29 by DavesCodeMusings

thomaslr, 2023-12-01:
don't know if this will help will help the original request but I also played with rshell which seems to be board neutral.

pip install rshell.  It "mounts" your board on /pyboard and allows file operations which are more like everyday cp rm than mpremote provides.
It also provides rsync which after doing some experimentation I can confirm only sends up changed files to your board.

rsync -m my_project_dir /pyboard 

hope it helps