DavesCodeMusings/mpremote-vscode #21: Single quotes not accepted by mpremote

somervda, 2024-03-13:
mpremote-vccode seems to wrap file paths etc in single quotes which do not seem to be accepted by mpremote. If I try the same commands with double quotes they will work. Maybe mpremote has changed to only accept double quotes?

For example when I try an use the file system ls command I get this error

C:\projects\owonpy>py.exe -m mpremote connect COM3 fs ls '/'
ls :'/'
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
TypeError: unsupported types for __truediv__: 'str', 'str'

BTW: I like the extension, it will be a handy addition when using mpremote so I am hoping this is an easy fix or there is some setting I should be using.

DavesCodeMusings, 2024-03-13:
Well, shoot. That's a new one. Single quotes have always worked up until now. Can you tell me what version of mpremote you're using? Maybe something's changed??? I'll try to reproduce the error.

DavesCodeMusings, 2024-03-13:
Looks like it's a shell thing.

If I use PowerShell, everything is fine...

```
PS C:\Users\Dave\src> py.exe -m mpremote connect COM9 fs ls '/'
ls :/
         964 boot.py
         101 config.py
           0 lib/
         461 main.py
```

If I use cmd.exe, I get an error similar to yours...

```
C:\Users\Dave>py.exe -m mpremote connect COM9 fs ls '/'
ls :'/'
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
TypeError: unsupported types for __truediv__: 'str', 'str'
```

Probably just change your default terminal to PowerShell in VSCode.
https://stackoverflow.com/questions/44435697/change-the-default-terminal-in-visual-studio-code

closed 2024-03-13 by DavesCodeMusings

somervda, 2024-03-13:
Thanks - that was my problem