DavesCodeMusings/mpremote-vscode #22: Serial Port Skip List

SteveKerle, 2024-03-21:
The serial port skip list does not appear to work correctly. On my system I add four entries, /dev/ttyS0,/dev/ttyS1,/dev/ttyS2,/dev/ttyS3. Only /dev/ttyS0 and /dev/ttyS2 are removed from the serial port list.

DavesCodeMusings, 2024-04-01:
I am not able to recreate this. I'm using Windows and I don't have enough serial ports. If anyone out there has a similar setup and can lend a hand, I'd appreciate any insights.

What I was able to do was verify the serial port skip list shows up properly in debugging info and that a port in the skip list does indeed get skipped. (See below.)

With _/dev/ttyS0,/dev/ttyS1,/dev/ttyS2,/dev/ttyS3_ in the skip list...
```
Calling mpremote as: py.exe -m mpremote
Creating new mpremote terminal.
mpremote version: 1.22.0
Detected serial ports: (1) [{…}]
Serial port skip list: (4) ['/dev/ttyS0', '/dev/ttyS1', '/dev/ttyS2', '/dev/ttyS3']
Avaiable serial ports: (1) [TreeItem]
arg1:
(1) [TreeItem]
0:
TreeItem {collapsibleState: 0, label: 'COM9'}
length:
1
```

With _COM9_ by itself in the skip list...
```
Calling mpremote as: py.exe -m mpremote
Creating new mpremote terminal.
mpremote version: 1.22.0
Detected serial ports: (1) [{…}]
Serial port skip list: (1) ['COM9']
Avaiable serial ports: (0) []
arg1:
(0) []
length:
0
```

I also tried various combinations like: _COM1,COM2,COM3,COM4_ ; _COM1,COM2,COM3,COM4,COM9,COM10_ and all results were consistent with expected behavior.

The only thing I can think of to suspect is hot-plugging USB devices. If you add or remove a device, you need to hover over _SERIAL PORTS_ in the VS Code Explorer pane (on the left side of VS Code) and click the Refresh icon (circular arrow thing.)

mhavill, 2024-04-02:
I am having the same issue as Steve on Windows 10 
I have 3 ports which are dedicated to my laptop subsystems COM3, COM4, COM5
However when I out "COM3, COM4, COM5" in the skip list, it will hide only COM4 and COM5
I am currently using COM15 for my microcontroller

DavesCodeMusings, 2024-04-02:
Just curious if folks are using spaces after commas or not.

If you are using spaces after the commas, try taking them out and see if that fixes things. It might be a whitespace problem. Just a guess.

mhavill, 2024-04-03:
> Sorry for taking a while to reply, only just returned from the UK and didn't have my laptop with me. I don't have much experience of Javascript but I would suspect the refresh function in PortListDataProvider. The comPortList is iterated over using a for loop. If found in comPortSkipList then that com port is removed using the splice function which will cause all following entries to be shifted in the list. The for loop will then increment causing the entry after one that has been removed by the splice function to be skipped from the check.

Hi Steve,Sounds like a good idea. For me trying to hide COM3, COM4 and COM5 it seems to leave COM3 regardless of its order in the list. Just any two in the list and both are hidden. BtW same result with and without spaces between entries. RegardsMick HavillMob +61 423 023 ***@***.***

DavesCodeMusings, 2024-04-03:
@SteveKerle ... No problem. Been on spring break myself and still catching up. Probably be a while before I can look at it.

SteveKerle, 2024-04-03:
I don't yet know how to build and test vscode extensions so am unable to test my suggestion but may I propose the code below

 `
    this.portList = [];
    for (let i=0; i<comPortList.length; i++) {
        if (comPortSkipList.includes(comPortList[i].path)) {
            comPortList.splice(i, 1);
        }
    }
    comPortList.forEach(port => {
        this.portList.push(new TreeItem(port.path));
    });
    console.debug('Avaiable serial ports:', this.portList);
`

is changed to something like this?

`
    this.portList = [];
    comPortList.forEach(port => {
        if (!comPortSkipList.includes(port.path)) {
            this.portList.push(new TreeItem(port.path));
        }
    });
    console.debug('Avaiable serial ports:', this.portList);
`

DavesCodeMusings, 2024-04-03:
@SteveKerle I started building extensions with the help of this dude's video: https://www.youtube.com/watch?v=q5V4T3o3CXE

DavesCodeMusings, 2024-04-20:
Sorry to take so long on this. Looks like @SteveKerle has identified the problem and the proposed solution should do the trick.

DavesCodeMusings, 2024-04-20:
Updating to version 1.21.10 should fix this bug. I only have one serial device and that's the ESP32, so I'm unable to test. Nothing is broken, so I'm closing the issue. Please open a new issue if the problem persists.

closed 2024-04-20 by DavesCodeMusings