DavesCodeMusings/mpremote-vscode #30: Enhancement: Able to use on files other than .py

ma261065, 2024-07-15:
I have a project which has some HTML assets (.htm, .css, .svg, .jpg, .css, .ico, .png) , and I can't use this extension to upload them to the device as it only shows in the context menu for .py files.

Could please you add an option to add a list of file types to the settings that will then show the mpremote context menu when selected?

DavesCodeMusings, 2024-07-15:
I can certainly look at what it would take to do this. Until then, a workaround would be to do a `sync` of the project's entire directory structure. I've used that trick to upload my HTML content (along with everything else.)

ma261065, 2024-07-26:
Thanks for taking a look. BTW, the sync option doesn't seem to work if there is a subfolder containing the HTML assets

DavesCodeMusings, 2024-07-26:
Thanks for including the output. I recently changed some things to support using a parent dir besides /. I wonder if I've broken something. If you downgrade your extension version to 1.21.10, that may solve it.

somervda, 2024-09-16:
Dropping back to 1.21.10 worked for me, I can sync my whole project again. Also an option to upload non py files would be a useful enhancement for me (I have a settings.json file). Thanks for a great VSCode extension but it may be worth withdrawing the 1.21.11 version.

DavesCodeMusings, 2024-09-18:
@somervda thanks for testing with the previous version. I tend to agree about rolling back the changes in 1.21.11. As well as introducing bugs, I don't think it's doing what was intended to do with the enhancement.

DavesCodeMusings, 2024-09-18:
1.21.12 is available now. It reverts the changes that caused the bugs.

somervda, 2024-09-18:
Thanks -I updated to 1.21.12 and is all going well. Thanks for the work on this extension.

DavesCodeMusings, 2025-02-20:
Feature added in 1.21.14. You can now upload any type of file using the context menu (right-click.) Previously you were restricted to .py files.

For example, upload a README.txt if you wish:

[image]

DavesCodeMusings, 2025-02-20:
If anyone using this feature could report back with your success / failure, it would be appreciated. Thanks!

somervda, 2025-02-20:
The new capability worked for me . I did a quick test with a settings.json file and it uploaded without a problem. Thanks Dave, I will be using the new feature in future. Sorry no active project right now so can't test more than that.

DavesCodeMusings, 2025-02-20:
> The new capability worked for me . I did a quick test with a settings.json file and it uploaded without a problem. Thanks Dave, I will be using the new feature in future. Sorry no active project right now so can't test more than that.

Appreciate the feedback. Glad it's working for you.

closed 2025-02-20 by DavesCodeMusings

ma261065, 2025-02-23:
Yep, also working fine here. Thanks - this makes my workflow so much smoother.