DavesCodeMusings/mpremote-vscode #31: Windows: VSCode extension looks for py.exe which does not exist and won't use python.exe or python3.exe

freemansoft, 2025-02-01:
The name or path of the Python executable should be configurable in settings.  This previous report had the right idea.
https://github.com/DavesCodeMusings/mpremote-vscode/issues/23

https://learn.microsoft.com/en-us/windows/python/faqs

> HOWEVER, the py.exe launcher will only work if you are using a version of Python installed from [python.org](https://www.python.org/downloads/windows/). When you install Python from the Microsoft Store, the py command is not included. For Linux, macOS, WSL and the Microsoft Store version of Python, you should use the python3 (or python3.7) command.

DavesCodeMusings, 2025-03-13:
Should be possible with a VS Code setting. `vscode.workspace.getConfiguration('mpremote').executableInvoke` or something similar. There's already a `getMPRemoteName()` function in utility.ts that returns 'py.exe -m mpremote' or 'mpremote', depending on the detected O.S. Having a check for a user defined setting prior to that bit should do it.

DavesCodeMusings, 2025-03-17:
Enhancement in commit https://github.com/DavesCodeMusings/mpremote-vscode/commit/2fb50cabb72557fb2b0dbe50c094836e46416f9a

closed 2025-03-17 by DavesCodeMusings