DavesCodeMusings/mpremote-vscode #32: mpremote now uploads files as '\main.py' instead of 'main.py'

SawDust58, 2025-02-19:
I have been using mpremote (latest version) in VS Code for over a year and it has worked wonderfully. However, now when I right click in a file editor window -> mpremote -> 'upload file to remote filesystem (cp)' my file 'main.py' gets copied as '\main.py'. I see this \filename in the terminal window during the copy, and I see the \filename on my device when I list files. This behavior occurs on all python files. When I 'upload all project files to remote filesystem (sync)' all files get copied with expected filenames (no \ char), and my device works fine.

Background: I recently dabbled with a Waveshare rp2040-Zero using VS Code and mpremote but had trouble seeing messages in Terminal (crashes). Then I downloaded PyCharm to check it out. Went back to VS Code and got the same rp2040 working using mpremote and Serial Monitor. Finally went back to my ESP32 and discovered mpremote file updates now include the'\' character in file names.

The settings in VS Code overwhelm me so I don't change them, and I can't find a setting that might change file names to add \ during copy, Online searches have not been helpful. I'm guessing nothing has changed in mpremote, so I would appreciate any help you can offer to figure out what has changed on my end.
 
[image]

Jeff

DavesCodeMusings, 2025-02-20:
I can replicate it... sort of. I see the backslash in the mpremote command to upload a file (cp), but it's not there when I list (ls) the files after.

```
PS C:\Users\dave\src\ESP\ThimbleTest> py.exe -m mpremote connect COM3 fs ls '/'
ls :/
         139 boot.py
PS C:\Users\dave\src\ESP\ThimbleTest> py.exe -m mpremote connect COM3 cp 'c:\Users\dave\src\ESP\ThimbleTest\config.py' ':\config.py'
cp c:\Users\dave\src\ESP\ThimbleTest\config.py :\config.py
PS C:\Users\dave\src\ESP\ThimbleTest> py.exe -m mpremote connect COM3 fs ls '/'
ls :/
         139 boot.py
          66 config.py
PS C:\Users\dave\src\ESP\ThimbleTest>
```

My guess is its a bug in the extension that's been there for a while and I just haven't noticed.

I'll have to dig into the code, but I suspect it's because the upload tries to maintain the same subdirectory structure as your project folder. For example, uploading C:\Project\lib\mymodule.py ought to end up as /lib/mymodule.py on the microcontroller. Somewhere in there it's using Windows style \ instead of MicroPython's /

SawDust58, 2025-02-20:
Thanks for the prompt reply, Dave.

 

I’ve had this same file structure for a couple years (long project) and I have not seen this problem until yesterday. I don’t remember if I updated MPRemote at that time.

 

When file name is ‘\main.py’ on my device, the device immediately crashes on restart, so I would have noticed it earlier if the problem has been lingering ‘for a while’. I have not noticed if the \ has been in the cp command line all this time because my device always worked.  Any chance installing PyCharm yesterday changed some file that Python or VS Code uses? PyCharm asked if I wanted to import VS Code settings (I said no), so I don’t know if there is some layering going on. The coincidence of that install and this problem is suspicious. Yesterday was also the first time I used VS Code Serial Monitor (Microsoft), and used MPRemote with a rp2040 device (not ESP device).

 

Let me know if you want me to try things on my end.

 

Regards,

Jeff

 

P.S. Originally I used PyMakr but struggled with the extension for various reasons. MPRemote has been rock-solid in all features, it is very simple to use, and it simply works. I wish I could upload json, html, and css files individually. Also, I wish I could upload py files individually to a subfolder. But I am very happy with MPRemote features as-is because the extension works flawlessly.

DavesCodeMusings, 2025-02-20:
Just published the fix on the marketplace. (v 1.21.13) When you have a chance, update your extension and try to reproduce the error. On my system, the backslashes are properly replaced with forward slashes now. Hopefully that will fix things going forward.

SawDust58, 2025-02-20:
I tested new version on two different files, your fix works!  I still see \ in cp command line as you pointed out, but most importantly my device directory is correct. Fabulous response time Dave, thank you!

[image]

SawDust58, 2025-02-20:
Addition to my previous msg: In case you get the itch to add new features, file extensions that I upload are py, json, txt, html, css, js, (png?)

DavesCodeMusings, 2025-02-20:
Glad it's working for you!

As for this enhancement...
> Addition to my previous msg: In case you get the itch to add new features, file extensions that I upload are py, json, txt, html, css, js, (png?)

It's already in issue #30. I'm going to close this one as done and track the enhancement for other file types in #30.

closed 2025-02-20 by DavesCodeMusings