DavesCodeMusings/mpremote-vscode #33: Option to set default port

chrismcsoup, 2025-03-09:
Hi,

thank you for the great extension! One suggestion, for me it would be helpful to be able to set a default port for example in a .mpremote config file in the project, so that we can run/re-run a file quicker without selecting the correct port every time.

DavesCodeMusings, 2025-03-10:
Take a look at the Serial Ports in the Explorer pane of VS Code. You can right-click the serial port of your device and select the command you want from the context-menu (see screenshot below.) This removes the need to specify the serial port, since the extension knows what you right-clicked. Also, if only one microcontroller is plugged in, it should default to that port already. At least, that's how it works for me on a Windows host with a single miscrocontroller.

[image]

chrismcsoup, 2025-03-16:
Thank you, I saw that option. I have multiple serial ports on macos that get's detected, so I always have to select the right one or have to find it in the serial ports view. 

It is not a huge issue, it is just one or two more clicks, which I wanted to save. But I'll probably configure the launch.json to execute mpremote for executing the python code from the active editor when pressing F5. 

Anyway, thank you for the great plugin and your help!

closed 2025-03-16 by chrismcsoup

DavesCodeMusings, 2025-03-16:
> I have multiple serial ports on macos that get's detected, so I always have to select the right one or have to find it in the serial ports view.

There is a comma separated serial port skip list you can find in the extension's settings. If you're consistently picking up a non-microcontroller device, you can use this to have it be ignored.

[image]

chrismcsoup, 2025-03-16:
Thank you. I must have overlooked that setting.