DavesCodeMusings/mpremote-vscode #37: Automatically save the currently edited file when selecting "Run File" from the context menu.

qianxuu, 2025-04-20:
Currently, when I right-click in the editor and select "Run/Upload File," if the current file has unsaved changes, these changes are not executed. I need to manually save the file first and then run it. This adds unnecessary steps, and sometimes I forget to save, resulting in running an older version of the code.

DavesCodeMusings, 2025-04-20:
I would suggest using VS Code's "auto save" to take care of keeping changes current in the file. I can't think of a compelling reason to duplicate this functionality in the extension when it's already built-in to VS Code.

qianxuu, 2025-04-20:
@DavesCodeMusings I was used to using vscode's f5 key debugging feature, which automatically saves the currently edited file. I don't really want to turn on vscode's global autosave, but you can also make run file an option in the debugger if you want.

DavesCodeMusings, 2025-04-20:
Going back in the commit history, it looks like I had the feature in there at one time. https://github.com/DavesCodeMusings/mpremote-vscode/commit/a691eb626967f1e86fd8a2449f4e27eaa9343fe6

I removed it later. I can't recall the exact reasoning, but I think it was problematic. Might have been it could only detect unsaved changes in the currently active editor and not other open editors. That's just a guess.

closed 2025-05-04 by qianxuu