DavesCodeMusings/mpremote-vscode #38: Windows. Error running command mpremote.*****

drwheeler, 2025-05-05:
Version 1.21.15 works fine.
Version 1.21.16 does not work at all.

SERIAL PORTS selection is always empty.
Rescan available serial ports or any other mpremote extension command produces an error:

Error running command mpremote.refreshSerialPorts: command 'mpremote.refreshSerialPorts' not found. This is likely caused by the extension that contributes mpremote.refreshSerialPorts.

Python 3.13.3
mpremote 1.25.0
pip      25.1.1
pyserial 3.5

DavesCodeMusings, 2025-05-05:
Version 1.21.16 is when I switched to using `mpremote devs` to get the list of attached devices. What happens when you run `py -m mpremote devs` from a PowerShell prompt? 

It should look something like this:

```
PS C:\Users\dave> py -m mpremote devs
COM3 0001 10c4:ea60 Silicon Labs None
```

DavesCodeMusings, 2025-05-05:
I have updated my Python, pip, and mpremote version to match yours and... you're not going to like this, but... I can't reproduce the error.

The only thing I can think of that's different is the Python virtual environment. I don't use a virtual environment with my setup. I'm guessing by the (.venv) in front of your PowerShell prompt, you might be???

Anyway, if that's the case, try checking the box to skip start-up checks in the extension's options. See screenshot below.

[image]

drwheeler, 2025-05-06:
I installed mpremote globally and that fixed the problem. No more issue. Thx.

closed 2025-05-06 by drwheeler

DavesCodeMusings, 2025-05-06:
Glad to hear it's working.